zs6buj/AntTracker #85: Antenna Track Collaboration

LZ-3D, 2024-09-23:
Hi, I'm Luca, I've been working on my antenna track project for several months, now that it's time to program it I came across your project.

So I'll show you my results, 3D modeling of the antenna.

For control I use an esp32 tft display, with gpio expansion and two drivers for nema17 with minimum and maximum limit switches.

the goal is to make the most of the MLRS telemetry for tracking, and capture the analog or digital video signal as best as possible
and send the telemetry to the PC with Mission Planner via wifi.

My project is not yet public but it will be soon.

now I try your code, I'd like to collaborate with you.

[image]

[image]

zs6buj, 2024-09-24:
Hi Luca, your project looks very nice and of course I would like to collaborate with you. You say you have limit switches on the steppers, so we might need more available gpio pins on the mpu. The moba library I use can set and remember an arbitrary "start" position within a session, so strictly speaking we don't need limit switches. Also, sometimes the shortest angular distance to the desired angle is sometimes the opposite direction past where the limit switch would be.

I look forward to hear from you.

LZ-3D, 2024-09-27:
I put the limit switches because if the belt slips on the pulley the motor is convinced that it is in a position that in reality is not exact, so if it presses the limit switch the max. and min. limit are calculated again.

for the gpio ports I use MCP23017 16 EXPANDER and for the display it is the one with esp32 the model is 2432S028 at the moment your code compiles correctly, in the next few days I will connect everything and try it, the project is continuously updated.
[image]
[image]
[image]

zs6buj, 2024-09-28:
Looks like you know what you need to do. I assume the two helical antennas are for video, and the Maxon is for control, maybe 900 MHz? Very nice. Let me know how it goes.

LZ-3D, 2024-10-09:
there are some protocols in your sketch that don't work, at the moment I abandoned the display with esp, because it has few gpio, so I redid everything as in the photo, I also eliminated the stepper motors, I'm waiting for the 30kg servos. I'll keep you updated.
[image]

zs6buj, 2024-10-10:
Thank you for the feedback. It would be useful if you could provide details and debugging data about which protocols don't work.

TomBeaucourt, 2024-11-18:
Hey @LZ-3D 
I could use some help. I'm really not comfortable with Visual Studio Code. I need to flash my Liygo TDisplay S3 with this firmware. Could you help me? :) (Here’s my Discord if needed: tom.rev.moe)

zs6buj, 2024-11-19:
Hi Tom, my apologies. We are moving house and I also don't have fibre connect up yet. MobaTools may not yet have support for some boards. The library has some nice features that I use in the tracker software. 

VSCode has a bit of a learning curve, but it is an all-round great IDE for cpp. You could post specific questions here and I will try to get back to you, but I can't promise immediate answers. By the way, you can also try posting questions on co-pilot. It is surprisingly perceptive.

TomBeaucourt, 2024-11-19:
Oh, thank you so much for your response! I apologize for the inconvenience. Yes, I ask questions to ChatGPT and Copilot to help me out. I also ordered the ESP32 Devkit that you mentioned in the code. At least, if I can't get it to work with the LilyGO, I’ll be confident that it will work with this one, as I need to deliver a functional project soon. I'll do my best to make it work. Thanks a lot for your help, and good luck with your move! 😊

TomBeaucourt, 2024-11-19:
Quick question: is it possible to switch ELRS targets using a button? For example, to allow a friend to use my station? And if yes, can the buttons also be used for other menus? For now, I haven’t seen any buttons in the 3D model. As for the amount of data to display, is a 1-inch screen sufficient, or should I use two?

zs6buj, 2024-11-19:
A 096 inch OLED display is enough to display basics, but a bigger display supports a HUD. Sure you could use a switch like you describe, by adding a bit of code.

TomBeaucourt, 2024-11-21:
Hello, how are you?
On my side, I’ve made a lot of progress! I now have functional telemetry, servo control, and even a display ;)
However, I’m encountering some issues with the servos. During the self-test, the angles are incorrect, and I exceed the physical limits of my structure. Of course, I stop the system before the servos are forced too much.

I’ve tried adjusting various values, disabling reverse settings, etc., but nothing seems to fix it. Maybe I’m missing a parameter?
I’m using 270-degree servos for azimuth, but for elevation, it’s not possible—it needs to be limited to only 150 degrees.

Do you have any advice or ideas?

zs6buj, 2024-11-22:
Hi Tom, nice progress!

Let's ponder the question of servo reach for a moment. Imagine the 360 degree field around the tracker is split into an arbitrary  front-field, the field in front where your UAV starts its flight, and the back-field, where it might possibly venture.

Consider way to point the antenna in the back-field, shaped like half a hemisphere.  If our azimuth servo can move 360 degrees, the elevation servo only ever needs to move 90 degrees. Indeed, it should never move more than 90 degrees. This is one possible working configuration.

However, if the azimuth servo moves 180 degrees, and the elevation servo also moves 180 degrees, to reach the back-field the elevation servo must flip the antenna over to that field. This is the second possible working configuration. 

Sometimes the physical servos are mounted such that their action is reversed. This is not a problem becuase we simply 'reverse the reverse' in software independently of what I said above.

In summary,  given your servos' physical reach, I would choose the second working configuration.

TomBeaucourt, 2024-11-22:
Hello, I tried again all day but nothing new. I understood the idea of both sides, but now my problem is that during the motor testing phase, the maximum limits are not being respected. Also, when I go into the motor file, the variables are not being taken into account—azMidFront and elStart aren't even recognized. I must be really bad at this, it's unbelievable :/

Have a nice day and thank you very much.
[image]
[image]
[image]
[image]

zs6buj, 2024-11-27:
G'day Tom, 

I suggest that you incrementally reduce maxAzPWM from the 2257 you see there, until the azimuth servo physically stops at 180 degees. Then also reduce the maxElPWM until that servo physically stops at 90 degrees. Try going down in (say) 10 microseconds at a time initially, then in smaller increments.

It might also be necessary to move the starting point of each servo by increasing minXXPWM, until you are properly in the working range of the servos, but first do the above. If you eventually change the minXXPWMs, then the maXXPWMs will likely have to change again. So you are revising iteratively.

[image]

TomBeaucourt, 2024-11-27:
Hello, thank you very much for the response. I will test all of this then. Is it normal for the variables to not be recognized in the file motors.CPP?

TomBeaucourt, 2025-01-01:
Hello @zs6buj and Happy New Year 2025! I hope the move is going well. I've made a lot of progress on the tracker; it's almost finished, but I have a last-minute bug. I'm having an issue with my buttons: one is supposed to navigate back to the startup menu with the list of executed commands, and the other sets the tracking. The only problem is that I get interference. Occasionally, when I bring my fingers close to the button, it triggers without me even pressing it.

Both buttons behave the same way. In terms of wiring, one side of the button is connected to a GPIO pin, and the other to GND. I've tried several things: adding capacitors, resistors, and even checking the code to ensure the debounce logic is implemented. At this point, I must admit I haven't found a solution—I’m still not much of an electronics expert! 😂 Even when I connect the two wires directly, it sometimes works, but other times it generates disturbances.

Thanks for developing the code and especially for taking the time to read my messages. Once my project is fully completed, I'll share the STEP version of it.

zs6buj, 2025-01-01:
Hi Tom, and a happy new year to you and your family! I look forward to seeing your final product. 

Some pins of the ESP32 Dev Board are "touch pins", i.e. they are ultra high impedance, and any charge lingering on your skin will trigger true. Note the cyan coloured labels in  this diagram and avoid using those pins with your push buttons in this application. If you must use a touch pin, you could try 10k pullup resistors from the pin to the 3.3v line.

[image]

TomBeaucourt, 2025-01-01:
Hello again, it's incredible—everything is working again! Thank you so much for the explanations. 😊 Now, I just have to adjust the final settings, and we're good to go! 🚀

TomBeaucourt, 2025-01-04:
Hello, I still have a few questions. I did some static tests this afternoon (with a wing I was holding because the weather doesn’t allow for flying yet). I was in GPS tracking mode using the drone’s GPS. I noticed that sometimes the elevation servo reaches its limit (it literally hits its mount). I then have to unplug everything to avoid breaking the servo. I think I haven’t yet found the correct settings for the servo limits.

Otherwise, during a test where I did a good calibration, I think everything went well. What would be the best tracking mode: GPS or using the compass of my flight controller (SpeedyBee F405 Wing)?

I also have one last question: I can’t get the RSSI to display as shown in the picture; it remains at 0%. Do you have any suggestions? I tried changing the settings on the ELRS backpack and checking if the sensor was properly detected, and everything seems fine on that side.

I feel like I’m getting close to the end, and I’m really looking forward to everything working perfectly. What bothers me the most at the moment are the servos. Thank you very much for the buttons; they work really well ^^. Have a great day, and thank you so much for your help and for developing the code.
[image]

zs6buj, 2025-01-05:
Hi Tom, try reducing the elevation max PWM by 100 and check the result. If you have gone too far, increase it again until you find the best value.

Tracking is derived from GPS coordinates. The compass heading is only used to relate the position of the tracker to real world coordinates. 

I would have to look at the code to remind myself about the rssi. Remember the rssi you want is that of the transmitter/RC controller signal as seen by the aircraft or UAV, not the espnow signal.

zs6buj, 2025-03-18:
Closing as stale. Please reopen if you need to

closed 2025-03-18 by zs6buj