zs6buj/AntTracker #86: V2.22.9 with ESP32 Dev module, WiFi, GPS and Compass can't be compiled

GauchoFPV, 2024-09-27:
Hi Eric,
I tried to compile the latest version but got several errors:

[image]
[image]

I could change WiFiUDP.h to WiFiUdp.h because I assume it got changed in the library but am not able to correct the other issues because I don't have any programming skills...

Could you help me please?
Here is my config.h:

[config.h.txt](https://github.com/user-attachments/files/17160094/config.h.txt)

Thank you and have a nice day

zs6buj, 2024-09-28:
GauchoFPV: Thanks for the library name fix. Updated in v2.22.11.

In your config.h you have:

```
    #define SSD1306_DISPLAY         // OLED display type
    /* Below please choose either Touch pin-pair or Digital pin-pair for display scrolling
      *  Pin == -1 means the pin-pair is not used
      */
    #define Pup           -1        // Board Button 1 to scroll the display up
    #define Pdn           -1        // Board Button 2 to scroll the display down
```
If you #define SSD1306_DISPLAY, then you must define pageup(Pup) and pagedown(Pdn) gpio pins to scroll the display. I have used your config setup in v2.22.11. and I'm getting a clean compile here now.

GauchoFPV, 2024-09-28:
Hi Eric,

thank you very much and shame on me!

I just found another small mistake in the antTrack.ino, Line 137 : #include <globals.h> --> #include "globals.h"

Now I can compile and upload it without  problems, thank you so much!!!

zs6buj, 2024-09-28:
Hi, It's best to use v2.22.11 if you can. But also v2.22.09 has globals.h included in the antTrack folder.
[image]

GauchoFPV, 2024-09-30:
Of course, I used 2.22.11 and globals.h was in the folder but Ibhad to replace  the <> with "" to make it work.

zs6buj, 2024-10-24:
Can we close this issue now?

GauchoFPV, 2024-10-24:
Sorry, I'm not used to GitHub but yes, the ticket can be closed. Thank you very much!!!

closed 2025-03-18 by zs6buj