zs6buj/AntTracker #89: V2.22.16 PIO error build

sosko71, 2025-01-01:
If I try to build last version for Lilygo-t-display this two error appear:

[{
	"resource": "/c:/Users/Silvo/Desktop/v2.22.16_AccelStepper/antTrack/src/main.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'BuiltinLed' was not declared in this scope",
	"startLineNumber": 507,
	"startColumn": 7,
	"endLineNumber": 507,
	"endColumn": 7
}]

and

[{
	"resource": "/c:/Users/Silvo/Desktop/v2.22.16_AccelStepper/antTrack/include/utilities.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'BuiltinLed' was not declared in this scope",
	"startLineNumber": 1049,
	"startColumn": 7,
	"endLineNumber": 1049,
	"endColumn": 7
}]

on V2.22.15 build complete without error

zs6buj, 2025-01-02:
Thank you for reporting this. Fixed.

sosko71, 2025-01-02:
sorry but now there is more errors

[{
	"resource": "/c:/Users/sosko/Desktop/v2.22.17_AccelStepper/antTrack/src/main.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'adjustButtonActive' was not declared in this scope",
	"startLineNumber": 1055,
	"startColumn": 7,
	"endLineNumber": 1055,
	"endColumn": 7
},{
	"resource": "/c:/Users/sosko/Desktop/v2.22.17_AccelStepper/antTrack/src/main.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'button_millis' was not declared in this scope",
	"startLineNumber": 1057,
	"startColumn": 21,
	"endLineNumber": 1057,
	"endColumn": 21
},{
	"resource": "/c:/Users/sosko/Desktop/v2.22.17_AccelStepper/antTrack/src/main.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'my_deg' was not declared in this scope",
	"startLineNumber": 1060,
	"startColumn": 7,
	"endLineNumber": 1060,
	"endColumn": 7
},{
	"resource": "/c:/Users/sosko/Desktop/v2.22.17_AccelStepper/antTrack/src/main.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'direction' was not declared in this scope",
	"startLineNumber": 1060,
	"startColumn": 17,
	"endLineNumber": 1060,
	"endColumn": 17
},{
	"resource": "/c:/Users/sosko/Desktop/v2.22.17_AccelStepper/antTrack/src/main.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'az_step' was not declared in this scope",
	"startLineNumber": 1063,
	"startColumn": 7,
	"endLineNumber": 1063,
	"endColumn": 7
},{
	"resource": "/c:/Users/sosko/Desktop/v2.22.17_AccelStepper/antTrack/src/main.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'steps_per_degree' was not declared in this scope",
	"startLineNumber": 1063,
	"startColumn": 26,
	"endLineNumber": 1063,
	"endColumn": 26
},{
	"resource": "/c:/Users/sosko/Desktop/v2.22.17_AccelStepper/antTrack/src/main.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'azStepper' was not declared in this scope",
	"startLineNumber": 1064,
	"startColumn": 7,
	"endLineNumber": 1064,
	"endColumn": 7
}]

zs6buj, 2025-01-02:
Thank you for the valuable feedback. v2.22.17b

sosko71, 2025-01-02:
Its working now thanks

closed 2025-01-04 by sosko71