zs6buj/AntTracker #90: Guidance

MaRcOsss1, 2025-01-26:
I’m Aryan J. Shah, working on a UAS project to develop a 2-DOF antenna tracker using ESP32 and ESP8266. I came across your AntTracker repository and would like to implement it for my setup, where the ESP32 sends GPS coordinates to the ESP8266 (controlling the servos for pan and tilt).

Could you guide me on how to adapt your code for my setup, including any specific libraries, configurations, or calibration steps?

zs6buj, 2025-01-27:
Hi Aryan, I will need more detail of your system. Flight controller, transmitter and receiver, chosen telemetry protocol. The more detail the better.

MaRcOsss1, 2025-01-27:
SIYI MK15, mavlink protocol. I want to make a basic antenna tracker such that the motors would pan and tilt the antenna in the direction of the GPS coordinated received.

happytm, 2025-01-27:
@MaRcOsss1 You can use more simpler tracker meant for Sun tracker by removing code for sun position and replacing it with GPS co-ordinates from your moving target.

https://github.com/happytm/SunTracker

Thanks.

MaRcOsss1, 2025-01-28:
So the data that is being received on to the ground station I just need to write scripts to extract GPS data out of all the data and once I get that, it's all done right?

zs6buj, 2025-01-30:
Aryan, I recommend that you spend some time and read the wiki, then ask specific questions.

zs6buj, 2025-03-18:
closing as stale

closed 2025-03-18 by zs6buj