zs6buj/AntTracker #91: Compile Errors in Beta v2.22.17_AccelStepper with PIO

samnikos7, 2025-01-29:
I am trying to compile the latest Beta Version with Visual Studio PIO for "default_envs = lilygo-t-display" with these settings in "config.h":

- #define MEDIUM_IN  4    // Bluetooth Low Energy (BLE4.2)- ESP32 only
- #define PROTOCOL 9     // CRFS
- #define HEADINGSOURCE  3     // 3=Trackerbox_Compass

and I get many errors.

[image]

Could you please help me??

zs6buj, 2025-01-30:
Yeah, I never went back to add BLE support for CRSF. It's now in v2.22.18, beta beta. Thank you for the feedback. I would value any testing you can do as my BLE testbed is not easy to get to right now.

samnikos7, 2025-01-30:
Thank you for your quick reply. I can now compile it with no errors.
One last think, with the same setup I get errors when I try to change the compass in "config.h" to:
- #define HMC5883L

I have no QMC5883L to check but only HMC5883L so if you could fix it I can build this setup in the afternoon.

zs6buj, 2025-01-31:
My pleasure. I added the HMC5883 libraries into v2.22.18 for you and it's compiling now.

samnikos7, 2025-01-31:
Thank you again for your time but I get the above compile error for #HMC5883L:

[image]

samnikos7, 2025-01-31:
I am so sorry, it's compiling with no errors.

zs6buj, 2025-01-31:
Relax, I'm happy to help:)

samnikos7, 2025-02-01:
Is there any telemetry simulator like this just for testing for elrs?
It is really hard to calibrate without a simulator! 
[image]

zs6buj, 2025-02-01:
I am not aware of one. Sorry

zs6buj, 2025-03-18:
Closing as stale

closed 2025-03-18 by zs6buj