zs6buj/AntTracker #93: Code Versions

ValkyrieDomi, 2025-02-12:
Hi, I'd like to ask, which version is supposed to be the most up to date/relevant (for ESP32)?

In source/arduino/stable there are V2.20.04 and V2.20.06
Then Beta with V2.22.14

Then archive, which has V2.22.14. 

Even tho I was able to successfully compile, I would like to clarify the versions that are meant to be used, so I don't need to comment further about some compile errors, I faced, as those versions could be irrelevant.

Thank you in advance for reply, and thank you for the code & work on this antenna tracker 👍

zs6buj, 2025-02-12:
Hi, v2.22.14 beta contains several fixes and improvements. Check the changelog.txt. Try this version, but test it before you use it in the field.

ValkyrieDomi, 2025-02-12:
Many thanks, I thought so, that the beta is the most relevant. 
Thus also the confusion with V.2.22.14 in archive and beta folders. 

Would be better make zip packages for the newest releases that only contains needed and relevant libs and codes?