zs6buj/AntTracker #95: LOG from Serial Monitor - check

ValkyrieDomi, 2025-02-19:
Hello, I would like to ask You about my LOG from serial monitor. 

_FIY: I am using ELRS Mavlink RC, the AntTracker listens as STA on UDP_REMOTEPORT = 14550; thus connecting to my TX Backpack module._ 

I am getting **hb_counts**, I would like to get explanation what sort of info is that.

Anyway, the most observed issue is a loop of: 

Good flight computer GPS lock
gpsGood:1 

//some telemetry received meanwhile...

Flight Computer GPS timeout *********
gpsGood:0 

//and over and over again. 

TBH, Now I am testing at home, thus I am not sure if this could be the issue, receiving sats/GPS in stationary conditions at home.

 (Ideally, I should test outside but it is too late for me - dark when I get home).

Also one note, with **#define PROTOCOL 0     // AUTO detect protocol**, I am getting _Unknown protocol:0, aborting ...._
So I set **Mavlink 2** to get it work.

[AAT 2.22.13 Arduino IDE Serial Monitor LOG.txt](https://github.com/user-attachments/files/18873491/AAT.2.22.13.Arduino.IDE.Serial.Monitor.LOG.txt)

Thank you in advance for review and help! Have a lovely day.

zs6buj, 2025-02-20:
Hello, fortunately the log tells me that you are successfully receiving  mavlink telemetry. The hb_count refers to the mavlink heartbeat, which is sent periodically by the flight controller to signal that the link is alive. Timeout occur when no good GPS message has been received within a few seconds. The timeout period is defined in config.h. It seems that somewhere there is a delay, but you could also try increasing the timeout.

ValkyrieDomi, 2025-02-20:
Thank you very much. I will try that. 

However, under ideal conditions, what sort of rates/updates interval should I observe in serial monitor? If the telemetry messages are coming in constant rate let's say 1/s , should I see it the same way in serial monitor or it's somehow limited?

My point is, in the log's timestamps you can see there are sometimes big delays. 2,3,5 and once I seen like 10 seconds delay. 

Might be the ELRS Mavlink inconsistent. In MWP for example I don't see any problems with incoming telemetry nor any timeouts.

If the timeout won't help, I will try to feed it with native CRSF.

Extra question: I know there are limits for servo motion but Is it possible to disable the logic with 180/180 flip over? Just In case I have sort of big antennas that I don't wanna flip over, therefore 180 Azimuth coverage 90 Ele. would be enough. 

Thank you.

zs6buj, 2025-02-20:
Usually you can change the frequency of mavlink messages/message groups in the ground station app, like MissionPlanner. You could start there.

However, since you are using ELRS, you could try CRSF over ELRS. Your choice of protocol is usually governed by factors unrelated to the tracker. Mavlink is a rich protocol suitable for long or short range control, including missions / waypoints. ELRS has so far been associated with Betaflight and FPV, though these overlap somewhat. 

The crux of the matter is that you need a GPS coordinate message to determine the UAV's position in three dimensional space. If you get them infrequently, the tracker will stand still between messages. This may not be an issue is the UAV is far and/or slow.

Yes, we can inhibit the flip-over to prevent inertia/momentum problems with large antennas. The Moba library includes an acceleration parameter which could help with large antennas..

ValkyrieDomi, 2025-02-20:
I am not using Mission Planner, since I use INAV, the ELRS Mavlink is only for RC control and telemetry.

I got information that INAV can't slow down the telemetry data.

However, I will use Mavlink inspector and see the quality and consistency of telemetry received. 

For the timeout, I have not found anything with search, in config.h

Do you know the exact name, or If it's elsewhere?

zs6buj, 2025-02-20:
I was using my mobile phone and working from memory, but I'm home now. Give me a minute

zs6buj, 2025-02-20:
const uint8_t timeout_secs = 12;    in globals.h

ValkyrieDomi, 2025-02-20:
Perfect, thank you. 

The default 12 sec. is precisely shown in the log timestamps.

ValkyrieDomi, 2025-02-20:
Hello again. 

I have done some testing. 

1) The telemetry from FC to TX is all good with the ELRS Mavlink - checked with EdgeTX log
2) The telemetry from ELRS TX Backpack to MP & QGC without any issues. Data as GPS updates at 5Hz - therefore, entire system from FC all the way to the transmission on 14550 is OK.

The problem seems to be the tracker itself. 

The data in Serial Monitor are weird. I am including another LOG with UDP WiFi record read, which seems to have consistent rate every 2 seconds.  

It takes long time, to see some **Mavlink in (#33) GPS Int** data. And when it does, the rate is as discussed above. Plus after some time the timeout but let´s put that out of equation, since the inconsistency is the real issue. 

You are the master of the code, so I don´t want to speculate on that part. I am just thinking if maybe the ESP32 is not able to process it? Or, there might be an issue somewhere in the code? 

There is also some ap_mavlink_version=3 ... Mavlink 3? what? 

20:46:59.294 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81  motArmed=0  ap_custom_mode=4  ap_system_status=3  ap_mavlink_version=3

[AAT 2.22.13 Arduino IDE Serial Monitor LOG #2.txt](https://github.com/user-attachments/files/18894168/AAT.2.22.13.Arduino.IDE.Serial.Monitor.LOG.2.txt)

zs6buj, 2025-02-20:
Yes, UDP read every 2s is definitely a read flag. It is almost too consistent. I will try to take closer look later today, but I regretfully I don't have test kit with me here.

zs6buj, 2025-02-21:
Notice how the mavlink prefix sequence number jumps from 43 to 141 to 235. There seems to be a problem with the content on the UDP packets. These sequence number are supported to increment by 1, and wrap around withing one byte, so 0 thru 255

> 20:46:29.298 -> mav2:  /03 B0 /seq=43	len=31	/FD 1F 00 00 2B 01 01 93 00 00   #147	52 00 00 00 64 02 00 00 FF 7F DC 00 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 13 /00 00 /00 00 00 00 00 00 00 00 
20:46:29.298 -> Raw: 03 B0 FD 1F 00 00 2B 01 01 93 00 00 52 00 00 00 64 02 00 00 FF 7F DC 00 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 

20:46:31.265 ->  UDP WiFi record read:
20:46:31.265 -> mav2:  /D2 57 /seq=141	len=28	/FD 1C 00 00 8D 01 01 1E 00 00   #30 	3B B0 17 00 FC AD 00 BD 88 C3 64 3B FD 40 19 3F 46 FE 3A 3C 31 7A 17 BC D7 40 9A BB /FF FF /13 00 00 00 00 00 00 00 
20:46:31.309 -> Raw: D2 57 FD 1C 00 00 8D 01 01 1E 00 00 3B B0 17 00 FC AD 00 BD 88 C3 64 3B FD 40 19 3F 46 FE 3A 3C 31 7A 17 BC D7 40 9A BB 

20:46:33.295 ->  UDP WiFi record read:
20:46:33.295 -> mav2:  /89 CD /seq=235	len=16	/FD 10 00 00 EB 01 01 31 00 00   #49 	B3 61 44 24 A3 A2 95 06 62 1D 03 00 48 90 AC 5C /00 00 /00 00 31 7A 17 BC D7 40 
20:46:33.295 -> Raw: 89 CD FD 10 00 00 EB 01 01 31 00 00 B3 61 44 24 A3 A2 95 06 62 1D 03 00 48 90 AC 5C 00 00 00 00 31 7A 17 BC D7 40 9A BB

zs6buj, 2025-02-21:
Could you try UDP Monitor, maybe on your phone

zs6buj, 2025-02-21:
[image]

ValkyrieDomi, 2025-02-21:
I see, interesting finding, then there is so much missing. I can absolutely try the UDP monitor in about 8 hours from now, after work. 

_Side_ note: I did some performance and wifi tests, with real download data etc. on the ESP32 to make sure it's not acting up. On that part, nominal behavior.__

ValkyrieDomi, 2025-02-21:
How much data do you need? On UDP monitor I am having normal rate, no issues, but I guess you need some samples?
The thing is, it is such a pain to export a log from the APP. 

So meanwhile I am looking for alternatives.

ValkyrieDomi, 2025-02-21:
[UDP Monitor Log.log](https://github.com/user-attachments/files/18912494/UDP.Monitor.Log.log)

There is some, that I was able to pull out from **PacketSender** on PC

It has "ASCII" and then at the end HEX follows. 
For the time duration and msg count, it´s almost 4Hz update.

zs6buj, 2025-02-21:
Auto protocol detect was added fpr uart input quite a few years ago. It may not work for other mediums. I'll check that later as it should not be pertinent her

zs6buj, 2025-02-22:
Could you please insert this debug code in mavlink.cpp, around line 250. It could be that I have a parsing problem. We are parsing for 0xoD start/stop byte. Are you using VSCode/PIO or Arduino IDE?

    len = udp_object.parsePacket();
    int UDP_count = len;
    if(UDP_count > 0) 
    {
```
      #if defined DEBUG_ALL || defined DEBUG_MAV_INPUT
        log.printf(" UDP packet read, len:%d:\n", len);
        PrintMavBuffer(&msg);
      #endif    
```
        while(UDP_count--)

zs6buj, 2025-02-22:
Maybe the parsing is taking too long, and we should change to

`WiFiUDP.read(buffer, len);`

zs6buj, 2025-02-22:
Don't bother with the debug above

ValkyrieDomi, 2025-02-22:
Roger that, absolutely no special need to bother with the Auto Detect, I was just curious.

I am using both VS and IDE, so I can try whatever you throw on me.

Therefore, based on the last reply, I will wait for further instructions.

ValkyrieDomi, 2025-02-22:
Nah, one thing is for sure and that's the fact I won't be able to come with a solution. Honestly, I am happy when I achieve very simple - primitive code related stuff. 

I am just a try hard, trying everything I can from what's available. That's why I was gathering data and doing some tests to provide more feedback what's going on.

I can only, unfor

ValkyrieDomi, 2025-02-22:
_Ehm... How to say...._

Maybe fixed? 😄 

I implemented this code:

```
//================================================================
#if ((MEDIUM_IN == 2) && (WIFI_PROTOCOL == 2))     // WiFi UDP
bool readParseMavUDP(mavlink_message_t* msgptr)  
{
    if (!wifiSuGood) return false;

    bool msgRcvd = false;
    mavlink_status_t _status;
    int len = udp_object.parsePacket();

    if (len > 0) {
        // Allocate a buffer to hold the entire UDP packet
        uint8_t buffer[len];
        int bytesRead = udp_object.read(buffer, len);

        // Process the entire packet
        for (int i = 0; i < bytesRead; i++) {
            if (mavlink_parse_char(MAVLINK_COMM_2, buffer[i], msgptr, &_status)) {
                msgRcvd = true;
                UDP_remoteIP = udp_object.remoteIP();  // Remember which remote client sent this packet
                PrintRemoteIP();

                if (!hb_heard_from) {
                    if (msgptr->msgid == MAVLINK_MSG_ID_HEARTBEAT) {
                        hb_heard_from      = true;
                        hb_system_id       = msgptr->sysid;
                        hb_comp_id         = msgptr->compid;
                        hb_seq_expected    = msgptr->seq + 1;
                        hb_last_heartbeat  = millis();
                    }
                } else {
                    if (msgptr->msgid == MAVLINK_MSG_ID_HEARTBEAT) {
                        hb_last_heartbeat = millis();
                    }
                    checkLinkErrors(msgptr);
                }
                break;  // Exit after the first complete MAVLink message is received
            }
        }
    }

    return msgRcvd;
}
#endif
```

And the sequence is way more consistent. PLUS - no GPS timeouts. LED is still on, GPS fix being received. 
Tho there are still some pkt lost, from the 0-255 sequence. See the log sample (captured for one 0-255 loop to see how much missing):

[ATT_LOG_Mavlink_Parse_fix_maybe.txt](https://github.com/user-attachments/files/18922219/ATT_LOG_Mavlink_Parse_fix_maybe.txt)

ValkyrieDomi, 2025-02-22:
> Maybe the parsing is taking too long, and we should change to
> 
> `WiFiUDP.read(buffer, len);`

Could you give me some more hints for this? So I can try for improvements? 

_Also as I stated above, I have a little suspicion on the cheapo $hit ESP32. The ESP8266 was faster, even tho it has slower CPU, but unfortunatelly I cannot try on the ESP8266 as when compiling I get error_  `WiFi or Bluetooth works only on an ESP32 board, rather select serial in`

zs6buj, 2025-02-22:
Yes, this looks good. This was the plan and the code looks good on my phone. It's midnight here so I'll check more thoroughly in the morning.

zs6buj, 2025-02-22:
Though the result is still poor. Keep trying.  !uneducated :)

ValkyrieDomi, 2025-02-22:
I was unable to reach any better improvement. 
Tho I wanted to test it out with XPlane and HITL.
All data seems to be nominal from XPlane, while monitoring the serial monitor. 

Unfortunately, the tracker does not move at all. Static home position and "push button" works, LED also indicates the tracker is ready. (Servos are good, as the Tracker test them upon boot.)
But I am unable to detect, what could be the issue in the code, you know it better.

zs6buj, 2025-02-22:
The servos will not move before the distance away is more than 4m. May I request that you switch to v2.22.18 Accel, so that we are on the same page

ValkyrieDomi, 2025-02-23:
Correct, It is doable in simulator as well as IRL. Even, I moved away more than 4 meters since it´s an airplane 😄 
Let´s switch to `AntTracker-master\source\pio\BETA\v2.22.18_AccelStepper` and test then....

ValkyrieDomi, 2025-02-23:
> The servos will not move before the distance away is more than 4m. May I request that you switch to v2.22.18 Accel, so that we are on the same page

Now we are on the same page. 

With `HEADINGSOURCE  2     // 2=Flight Computer Compass` there is no sequence for setting home location. 👍

ValkyrieDomi, 2025-02-23:
`HEADINGSOURCE  1     // 1=Flight Computer GPS`

Clearly valid GPS,  Heading and distance (36m should be enough to get the vector, which it did)

[image]

LOG from the HITL flight (tracker side):

[HITL AAT Test LOG.txt](https://github.com/user-attachments/files/18928950/HITL.AAT.Test.LOG.txt)

Nope, tracker does not move servos -TRACKS- either in this mode. 
Either in this version, and even more than 4 meters. 

All it can do is initialize the `TEST_MOTORS`

zs6buj, 2025-02-23:
Apologies, I'll try to get back to you on this I'm tied up with family stuff. In the meantime I'm sorting out esp8266 so you can test with that.

ValkyrieDomi, 2025-02-23:
> Apologies, I'll try to get back to you on this I'm tied up with family stuff. In the meantime I'm sorting out esp8266 so you can test with that.

No worries. 
Also, If my last ESP8266 randomly burns down, I will end up with ESP32 anyway. If the code does not initiate the actual tracking, ESP8266 support is "not needed" atm. 

I went all the way up to the point where it should start tracking. I seen few RARE videos of the tracker in action, I just wonder how those people made it working ¯\_(ツ)_/¯

zs6buj, 2025-02-23:
So in v2.22.14  2024-10-24  we made this change:  Remove minAltAg completely as it hampers testing. 

Prior to v2.22.14 I inhibited tracking until altitude also was < min alt above ground = 4m. It would be best to go to v2.22.19 just published. Sorry for the tribulations. Your feedback has been valuable

ValkyrieDomi, 2025-02-23:
> So in v2.22.14 2024-10-24 we made this change: Remove minAltAg completely as it hampers testing.
>
> Prior to v2.22.14 I inhibited tracking until altitude also was < min alt above ground = 4m. It would [...]

If the **v2.22.19** was about to make **ESP8266** to compile, then success. The seqID seems almost perfect now, very little packets lost.

And there it ends. No Tracking, 

`HEADINGSOURCE  1     // 1=Flight Computer GPS`

```
Client connected: Remote UDP IP: 10.0.0.1  Remote  UDP port: 14550
 hb_count=1
Motors Armed!
 hb_count=2
 hb_count=3
Static home location AUTO set to Lat = 60.8223228 Lon = 11.0634298 Alt = -1.1
To get heading, carry craft straight ahead 10m, put down, then return and push tracker home button

Good flight computer GPS lock
 hb_count=1
 hb_count=2
 hb_count=3
Heading calculated = 90.0
Static home location set to Lat = 60.8223228 Lon = 11.0635157 Alt = -1 Hdg = 86 Alt = -1
Ending set_home phase
 hb_count=1
 hb_count=1
```

For

`HEADINGSOURCE  2     // 2=Flight Computer Compass`

No Home initialization, just hb_counts going and tele....

zs6buj, 2025-02-24:
Ok, so it looks like your esp32 is suspect, or all esp32s too slow. likely the former.

I'll look at the HEADINGSOURCE == 2  case, and see what's happening. I put together a quick udp send mule for testing, but now can't find an esp board to test with. I'll keep looking

zs6buj, 2025-02-24:
Can I ask you to build and test with these debug macros please 

```
//#define DEBUG_MAV_INPUT
//#define DEBUG_MAV_BUFFER  
#define DEBUG_AZEL
#define DEBUG_MOTORS 
#define DEBUG_MAV_HEARTBEAT 
#define DEBUG_MAV_GPS   
#define DEBUG_HOME
```

ValkyrieDomi, 2025-02-24:
It can be. Don't have a spare ESP32 to confirm. 

Will test the new release with the debugs when i get home.

ValkyrieDomi, 2025-02-24:
> Can I ask you to build and test with these debug macros please
> 
> ```
> //#define DEBUG_MAV_INPUT
> //#define DEBUG_MAV_BUFFER  
> #define DEBUG_AZEL
> #define DEBUG_MOTORS 
> #define DEBUG_MAV_HEARTBEAT 
> #define DEBUG_MAV_GPS   
> #define DEBUG_HOME
> ```

So with these debugs and the newest release + `HEADINGSOURCE  1`:

_Mavlink in #24 GPS_RAW_INT: ap24_fixtype=3 3D Lock  sats visible=12  GPS status=123  latitude=60.8223360  longitude=11.0627640  gps alt amsl=222  eph (hdop)=100  epv (vdop)=100  vel=0.0  cog=342.5
Mavlink in #33 GPS Int:  ap33_lat=60.822336 ap33_lon=11.062764 ap33_amsl=222 ap33_alt_ag=0.0 ap33_vx=0.2 ap33_vy=0.3 ap33_vz=0.0 ap33_hdg=150.4
  firstHomeStored:0  home.lon:11.062764  home.lat:60.822334  home.alt:0.0  home.hdg:0.0
Static home location AUTO set to Lat = 60.8223343 Lon = 11.0627642 Alt = 0.0
To get heading, carry craft straight ahead 10m, put down, then return and push tracker home button
Good flight computer GPS lock

GPS tele bla bla bla

Motors Armed!
Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=209  motArmed=1  ap_custom_mode=4  ap_system_status=4  ap_mavlink_version=3
 hb_count=1
Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=209  motArmed=1  ap_custom_mode=4  ap_system_status=4  ap_mavlink_version=3
 hb_count=2
Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=209  motArmed=1  ap_custom_mode=4  ap_system_status=4  ap_mavlink_version=3
 hb_count=3

Heading calculated = 153.7
Static home location set to Lat = 60.8221703 Lon = 11.0629358 Alt = -0 Hdg = 154 Alt = -0
  firstHomeStored:1  home.lon:11.062936  home.lat:60.822170  home.alt:-0.5  home.hdg:154.0
Ending set_home phase_

For `HEADINGSOURCE  2` as usual.

zs6buj, 2025-02-25:
I have added additional debugging in v2.22.20. It is configured for nodemcuv2 (esp8266). I think I might have found a problem with the mavlink/wifi combination, so it might be fixed. Thanks for you help. It's been great

ValkyrieDomi, 2025-02-25:
Great. I will try again when the servos - tracking is fixed.

ValkyrieDomi, 2025-02-26:
> I have added additional debugging in v2.22.20. It is configured for nodemcuv2 (esp8266). I think I might have found a problem with the mavlink/wifi combination, so it might be fixed. Thanks for you help. It's been great

`HEADINGSOURCE  2` terminal goes crazy with these messages:

[image]

`HEADINGSOURCE  1` here the tracking actually works when feeding it from X-Plane.
(still huge...enormous and rapid flow in terminal so I will reduce the sample here):

```
headingsource:1  hdgGood:0 sh_armFlag:0
headingsource:1  hdgGood:0 sh_armFlag:0
headingsource:1  hdgGood:0 sh_armFlag:0
headingsource:1  hdgGood:0 sh_armFlag:0

too many

Mavlink in #33 GPS Int:  ap33_lat=60.821914 ap33_lon=11.061852 ap33_amsl=241 ap33_alt_ag=19.0 ap33_vx=34.9 ap33_vy=-3.7 ap33_vz=-0.2 ap33_hdg=355.4
Heading calculated = 329.0
Static home location set to Lat = 60.8219147 Lon = 11.0618525 Alt = 19 Hdg = 355 Alt = 19
  firstHomeStored:1  home.lon:11.061852  home.lat:60.821915  home.alt:19.0  home.hdg:355.0Ending set_home phase
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:1 dist:0 minDist distTest:0
hom.lat:60.8219147 hom.lon:11.0618525 hom.alt:19 cur.lat:60.8219147 cur.lon:11.0618525 cur.alt:19 cur.alt_ag:0
 (dist:0 alt_ag:0
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:0 minDist distTest:0
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:0 minDist distTest:0
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:0 minDist distTest:0
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:0 minDist distTest:0

too many

Mavlink in #24 GPS_RAW_INT: ap24_fixtype=3 3D Lock  sats visible=12  GPS status=123  latitude=60.8221696  longitude=11.0618440  gps alt amsl=241  eph (hdop)=100  epv (vdop)=100  vel=35.2  cog=5.5
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:0 minDist distTest:0
Mavlink in #33 GPS Int:  ap33_lat=60.822170 ap33_lon=11.061844 ap33_amsl=241 ap33_alt_ag=19.0 ap33_vx=35.0 ap33_vy=2.3 ap33_vz=-0.2 ap33_hdg=5.4
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:1 dist:0 minDist distTest:0
hom.lat:60.8219147 hom.lon:11.0618525 hom.alt:19 cur.lat:60.8221703 cur.lon:11.0618439 cur.alt:19 cur.alt_ag:0
  hc_vector.az:359.03  hc_vector.el:0.00  hc_vector.dist:27  elapsed:28.706

too many

telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:27 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:27 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:27 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:27 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:27 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:27 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:27 minDist distTest:1
Mavlink in #24 GPS_RAW_INT: ap24_fixtype=3 3D Lock  sats visible=12  GPS status=123  latitude=60.8224192  longitude=11.0619520  gps alt amsl=241  eph (hdop)=100  epv (vdop)=100  vel=35.2  cog=18.3
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:27 minDist distTest:1
Mavlink in #33 GPS Int:  ap33_lat=60.822419 ap33_lon=11.061952 ap33_amsl=241 ap33_alt_ag=19.0 ap33_vx=33.5 ap33_vy=10.9 ap33_vz=-0.2 ap33_hdg=19.2
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:1 dist:27 minDist distTest:1
hom.lat:60.8219147 hom.lon:11.0618525 hom.alt:19 cur.lat:60.8224182 cur.lon:11.0619516 cur.alt:19 cur.alt_ag:0
  hc_vector.az:5.55  hc_vector.el:0.00  hc_vector.dist:55  elapsed:29.616
worldAz: 5       ourEl: 0        boxHdg: 355
ourAz: 100 ourEl: 0 boxHdg: 355
moveMotors az=100  el=0
az reversed Servo write az=80  el=0
```

ValkyrieDomi, 2025-02-26:
When tracker is tracking, `HEADINGSOURCE  1`  ,rapid flow of these messages:

```
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:136 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:136 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:136 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:136 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:136 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:136 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:136 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:136 minDist distTest:1
telemGood:1 gpsGood:1 packetGood():1 new_GPS_data:0 dist:136 minDist distTest:1
```

For all the examples also above, that would be several thousands of lines LOG in a very short time 😄

ValkyrieDomi, 2025-02-26:
With the current version I got 192 msgs out of 255. Improvement! 

From Tracking, sometimes significant delays:

```
ourAz: 76 ourEl: 18 boxHdg: 150
moveMotors az=76  el=18
az reversed Servo write az=104  el=18
hom.lat:60.8222847 hom.lon:11.0628204 hom.alt:0 cur.lat:60.8183022 cur.lon:11.0702419 cur.alt:198 cur.alt_ag:198
  hc_vector.az:137.79  hc_vector.el:18.30  hc_vector.dist:598  elapsed:256.732
worldAz: 137     ourEl: 18       boxHdg: 150
ourAz: 77 ourEl: 18 boxHdg: 150
moveMotors az=77  el=18
az reversed Servo write az=103  el=18
 hb_count=1
hom.lat:60.8222847 hom.lon:11.0628204 hom.alt:0 cur.lat:60.8165131 cur.lon:11.0687523 cur.alt:199 cur.alt_ag:199
  hc_vector.az:153.39  hc_vector.el:15.42  hc_vector.dist:717  elapsed:262.232
worldAz: 153     ourEl: 15       boxHdg: 150
ourAz: 93 ourEl: 15 boxHdg: 150
moveMotors az=93  el=15
az reversed Servo write az=87  el=15

```

zs6buj, 2025-02-26:
Ok, got it. We can remove unwanted debugging now, yeah?
Can I assume you are using the esp8266? 192 msgs out of 255 is not great.
Anyway, maybe we can concentrate on moving the antenna, and getting the acceleration sorted out.

ValkyrieDomi, 2025-02-27:
Yes. And Yes, now I am using ESP8266.
Sounds like a good plan.

ValkyrieDomi, 2025-02-27:
One quick question, before I get some new parts. 
Is there some specific OLED module that is preferred by the code? I came across some forum they were discussing some modules with specific driver not working. 

And, 360 servo option is meant continuous rotation or positional like sail winch servo? If continuous, I guess GPS w/ compass is mandatory for feedback loop (and slip ring)...

zs6buj, 2025-02-27:
[These](https://www.aliexpress.com/item/1005007614149117.html?src=google&pdp_npi=4%40dis!AUD!6.17!3.69!!!!!%40!12000041505921233!ppc!!!&src=google&albch=shopping&acnt=742-864-1166&isdl=y&slnk=&plac=&mtctp=&albbt=Google_7_shopping&aff_platform=google&aff_short_key=UneMJZVf&gclsrc=aw.ds&&albagn=888888&&ds_e_adid=&ds_e_matchtype=&ds_e_device=c&ds_e_network=x&ds_e_product_group_id=&ds_e_product_id=en1005007614149117&ds_e_product_merchant_id=5407422047&ds_e_product_country=AU&ds_e_product_language=en&ds_e_product_channel=online&ds_e_product_store_id=&ds_url_v=2&albcp=21819463808&albag=&isSmbAutoCall=false&needSmbHouyi=false&gad_source=1&gclid=CjwKCAiAlPu9BhAjEiwA5NDSAzPoX1t20zkmbcEY74fRyLEDES10THlxQYkTQsy74-B62kUThP-u_RoC4gMQAvD_BwE) display boards are cheap and sufficient for what you need. The ILI9341 based colour displays are nice if you want a HUD and a larger screen. They come in various sizes up to 5 or even 7 inches, but get expensive.

They work fine, but the support library TFT_eSPI must be setup properly for the board

zs6buj, 2025-03-05:
I've separated the UDP loop and parse loop for Mavlink as we previously discussed, and It's made a big improvement. I also added a profiling loop to test loop period between successful UDP reads. Then I pushed UDP packets at the tracker with100 mS (10Hz) rate, and it seems to cope. Let me know how your tests go with v2.22.21.

BTW, I also got some ESP32 Devkits fro Ali quick quick, and also ESP32C3 Supermini boards, and added support for them. They are quite cheap.

ValkyrieDomi, 2025-03-05:
Sounds very good and gonna test it out!

However, some compiling related stuff. When compiling with the OLED enabled, 

```
#if defined DISPLAY_PRESENT
    upButton.loop();
    dnButton.loop();
  #endif
```

**src\main.cpp:1071:5: error: 'upButton' was not declared in this scope;
src\main.cpp:1072:5: error: 'dnButton' was not declared in this scope;**

So I commented it out to proceed.

Next error:

**include/utilities.cpp:2030:5: error: expected primary-expression before '}' token** for void `handleDisplayButtons` at 1992 line.
`else` is misplaced before `#endif` at 2030.

Fixed on my end, compilable and OLED up and running...

Now let´s test 🥳

ValkyrieDomi, 2025-03-05:
End of testing:

```
Starting main.cpp
Version:2.22.21
Setting up Wire I2C: SDA:4, SCL:5
Display support activated: Landscape
64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
Target Board = 4     
ESP8266 / Variant is Lonlin Node MCU 12F
Expecting UDP In
Selected protocol is Mavlink 2
Expecting SERVO motors
headingsource:1 FC GPS
midFront is assumed to be 90 deg
Starting set_home phase
WiFi mode set to STA sucessfully
Trying to connect to ExpressLRS TX Backpack B5596C....
WiFi connected!
Local IP address: 10.0.0.100
WiFi RSSI:-60 dBm
Begin UDP using read port:14555  send port:14550   
UDP instance started, listening on local port 14555
end of setup()
moveMotors az=90  el=0
az reversed Servo write az=90  el=0
```

No Mavlink msg received. Here it literally ends.

ValkyrieDomi, 2025-03-05:
https://github.com/user-attachments/assets/26b5fed0-7dbe-414e-8f0c-41731e726bfe

Just to be sure, I´ve made Mavlink counter and with **average per minute**. So ESP8266 still works xD

zs6buj, 2025-03-05:
In my testing I switched the udp ports around:

```
Begin UDP using read port:14555  send port:14550   
UDP instance started, listening on local port 14555
```

You should switch them around. Sorry.

As far as acceleration goes, it is supported by the Moba library, but that library is getting out of date and does not support ESP32Cx or ESP32Sx, so I switched to Accelstepper. The problem is Accelstepper is for Steppers. So for Servos, we need to go back to Moba, or annd selection macro for motor type

ValkyrieDomi, 2025-03-06:
Thank you :) 

Ah I see, no worries, the lines that does not change normally I don't double check. Will test when home.

Yeah I was thinking to get steppers, but first I wanted some simple tracking solution. 

BTW: 
_I saw you used NEMAs with TB6600 drivers. The drivers are quite big IMHO for my needs. Something as small as possible is my Idea. I have old board from Ender 3 with drivers, so not entirely sure if the tracker is compatible with basically any stepper drivers?_ 

_Have you measured what current they take when loaded and tracking? (@12V) I am not entirely sure, how "big" such a setup would end up. With I suppose at least 4S battery and some *guessing* higher amp BEC/Step down._

zs6buj, 2025-03-06:
Yeah, I should move the unused Mavlink message types. I cut the code from another project. I'm working on selecting the motor libraries according to MCU. You would be better off with steppers IMO. Yes, I use Nema17 on other projects, not on the tracker. All my early trackers used servos. And sure you can use other stepper types, 2 wire, 4 wire. The mechanicals you need to research a bit.

zs6buj, 2025-03-06:
I use msg type #74 info for the HUD

ValkyrieDomi, 2025-03-07:
> Yeah, I should move the unused Mavlink message types. I cut the code from another project. I'm working on selecting the motor libraries according to MCU. You would be better off with steppers IMO. [...]

Ah, alright. Great, thank you :)

ValkyrieDomi, 2025-03-09:
I have a question about `#define AZ_SERVO_360`

If I convert a regular servo to 360, with multiturn pot, geared, to provide feedback, so it covers 0-359, Is it a viable option? Thus with this setting `#define AZ_SERVO_360` if I will loiter above AAT, when the servo reach 359, will it turn back - around to 0 and then continues? Similar to 180 flip logic? You know what I mean.

zs6buj, 2025-03-11:
Yes, I imagine this might work provided that the potentiometer wiper read is really accurate and consistent. The ADC resolution of the analog read pin would also play a part. For this application it should be enough. I successfully use a small IR switch board on another project to detect a starting point.

ValkyrieDomi, 2025-03-11:
[image]

Don't have the correct hall sensor for 360 angle application. But I have one of these, which I could try geared 1:1.

zs6buj, 2025-03-11:
A freely rotatable servo should be ok one you have a reference point, but I have not tested this. A stepper is definitely ok once you have a reference.

[28 more replies]

closed 2025-03-26 by zs6buj