zs6buj/AntTracker #96: V2.22.23 compilation error

HuangYing137, 2025-03-19:
Hi! Hello
When I use V2.22.23 and select Bluetooth Classic, a compilation error occurs

[image]

[image]

[image]

zs6buj, 2025-03-20:
Thank you for the feedback. This is a surprise! BluetoothSerial.h has a macro defined as  **bool setPin(const char *pin);**, so setPin is overloaded

```
src/config.h:537:31: error: expected unqualified-id before numeric constant
     #define setPin            12  // black
                               ^~
C:/Users/erics/.platformio/packages/framework-arduinoespressif32/libraries/BluetoothSerial/src/BluetoothSerial.h:63:14: note: in expansion of macro 'setPin'
         bool setPin(const char *pin);
```
              ^~~~~~

fixed in BETA 2.22.26

HuangYing137, 2025-03-20:
There is another problem. Here, the automatic protocol judgment is made, MEDIUM, I think you should use its MEDIUM_IN. At present, when automatically judging the compilation, it always forces an error here!

[image]

zs6buj, 2025-03-21:
Got it, thanks!

closed 2025-07-22 by zs6buj