zs6buj/AntTracker #97: PIO Compiling ERROR

samnikos7, 2025-03-30:
Hello again, I'm trying to compile the latest firmware for my lilygo-t-display with these parameters and fails:

- #define MEDIUM_IN  5    // ESPNOW - ESP32 and CRSF protocol only
- #define PROTOCOL 9     // CRFS
- #define HEADINGSOURCE  4     // 4=Trackerbox_GPS_And_Compass
- #define HMC5883L

[image]

I think the libraries for HMC5883L are missing! 

Could you please include them from now on to all your upcoming releases??

zs6buj, 2025-03-31:
Hello, I have added the library and fixed the error, but now have a new linking error. Give me a day or two to get back to this.

zs6buj, 2025-04-01:
When LilyGo released the t-display-s3, PlatformIO dropped support for the vanilla t-display for whatever reason. So I had to create our own lilygo-t-display.json, and put it in ../boards/ folder. The one I prepared before needed a small change. Could you let me know how you go with this.

samnikos7, 2025-04-01:
It's compiling now! I will test it later and post the results...

closed 2025-07-22 by zs6buj