zs6buj/AntTracker #98: Servo tests with 360 servos

toofen, 2025-04-18:
Hello, after taking a quite long break i am at it again. I have my tracker kind of working. I have a setup that requires me to have the azimuth servo moving 360 degress the problem i now have is that the elevator servo doesn´t move when i have TEST-servos enabled. The azimuth servo moves like it should but no movement from the elevator servo. The servo works if i select 180 degree mode. 

[image]

zs6buj, 2025-04-18:
Hello, your project looks nice. Could you please confirm version and post your config.h. You could try a debug session. Look in the debug.h file and uncomment appropriate lines. Then flash and test. I'm traveling and on my phone :)

toofen, 2025-04-19:
//================================================================================================= 
//================================================================================================= 
//
//                                    C O N F I G U R A T I O N 

#define MAJOR_VERSION       2
#define MINOR_VERSION      22
#define PATCH_LEVEL        28

//=============================================================================================
//================== Please select your options below before compiling ========================
//=============================================================================================

#define DEVICE_SYSID     251                     // Our Mavlink Identity - APM FC is 1, Mission Planner is 255, QGC default is 0 
#define DEVICE_COMPID    MAV_COMP_ID_PERIPHERAL  // 158 Generic autopilot peripheral - APM FC is 1, MP is 190, QGC is  https://mavlink.io/en/messages/common.html

//=============================================================================================
//============================    D E T E C T   M C U   F A M I L Y   =========================
//=============================================================================================
//                Don't change anything here
//
#if defined (__MK20DX128__) || defined(__MK20DX256__)
  #define TEENSY3X  
  #define TARGET_FAMILY   1      // Teensy 3.1 and 3.2  
  #define TEENSY_VARIANT 1    
#endif  
#if defined(__IMXRT1052__) || defined(__IMXRT1062__)
  #define TEENSY4X  
  #define TARGET_FAMILY   1      // Teensy 4.2  
  #define TEENSY_VARIANT 2  
#endif    
// Using official stmicroelectronics lib: https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json 
#if defined (STM32F1xx)   // in boards.txt / build.variant, like GenF1.menu.pnum.BLUEPILL_F103C8.build.variant=STM32F1xx/F103C8T_F103CB(T-U)
  #define TARGET_FAMILY   2      // Blue Pill STM32F103C8 
  #define STM32_VARIANT  1
#endif    
#if defined ESP32
  #define TARGET_FAMILY   3      // Espressif ESP32 
#endif    
#if defined ESP8266
  #define TARGET_FAMILY   4      // Espressif ESP8266
#endif 
#if defined(CONFIG_IDF_TARGET_ESP32C3)
  #define ESP32CX 3
#elif CONFIG_IDF_TARGET_ESP32C6 
#define ESP32CX  6   
#elif CONFIG_IDF_TARGET_ESP32S2
#define ESP32Sx  2        
#elif CONFIG_IDF_TARGET_ESP32S3
#define ESP32Sx 3   
#elif CONFIG_IDF_TARGET_ESP32S6
#define ESP32Sx 6        
#elif CONFIG_IDF_TARGET_ESP32H2
  #define ESP32Hx 2   
#endif

//=============================================================================================
//========================      S E L E C T   M O T O R   T Y P E      ========================
//=============================================================================================
 
// Select only one motor type
//#define STEPPERS 
#define SERVOS

//=================================================================================================                            
//===========================     S E L E C T   B O A R D   V A R I A N T     =====================  
//=================================================================================================

// Board is derived from board selected in IDE, variant is selected here
// Select only 1 variant
[... 1298 lines cut ...]
  #elif (MEDIUM_IN == 3) // BLUETOOTH Serial
        BluetoothSerial inSerial;
  #endif
#endif
#if ((PROTOCOL == 8) || (PROTOCOL == 0)) || (HEADINGSOURCE == 4) // NMEA GPS or Box GPS
  #include <TinyGPS++.h>
  struct compdate 
  {
    uint16_t yyyy;  
    uint8_t mm, dd, h, m, s;
  };
struct compdate dt = {0,0,0,0,0,0};
#endif

#if (HEADINGSOURCE == 4)
  #if (defined ESP8266) // Always SoftwareSerial
  #ifndef SWSERIAL_INCLUDED
  #include <SoftwareSerial.h>
  #define SWSERIAL_INCLUDED
        SoftwareSerial boxgpsSerial(boxgps_rxPin, boxgps_txPin);
  #endif
  #else
    #define boxgpsSerial Serial2 // Tracker box GPS
  #endif
#endif

toofen, 2025-04-19:
So its the latest version and the serial monitor seems to indicate that the elevator servo should move. I dont know if its just when testing the servos that the servo doesnt move, i have not come as far as testing it for real yet.

zs6buj, 2025-04-19:
```
const int16_t minEl = 180; // El lower limit in degrees, horizontal
const int16_t maxEl = 90; // El upper limit in degrees, straight up
```

This could be your problem. minEl should probably be 0, since in our conceptual model, horizontal is regarded as 0 degrees.

toofen, 2025-04-20:
That was it, now the servos both move, now i just have to get the geometry and degrees settings right.

closed 2025-04-20 by toofen