zs6buj/AntTracker #99: No signal when selecting Stepper Motors

dimircc, 2025-05-07:
Hi,
Your project and the work you have done is awesome. 

I successfully compiled for ESP32 Devkit V1, connected telemetry via ELRS Backpack and tested INAV in Simulator. 

When I configure for Servos everything is fine. But when i configure for Stepper Motors i get no Step and Dir signals (tested with osciloscope). At the same time the debug console is showing that the motors should move. 

Also when i press the adjust button before setting Home Position the MCU reboots, dont know if it relates to the same problem.

I attatched the config file here. [config.txt](https://github.com/user-attachments/files/20093669/config.txt)

Br
Dimi

Debug console :
`hom.lat:47.4841461 hom.lon:9.5693798 hom.alt:-1 cur.lat:47.4828911 cur.lon:9.5678635 cur.alt:66 cur.alt_ag:67
  hc_vector.az:219.17  hc_vector.el:20.38  hc_vector.dist:180  elapsed:422.911
worldAz: 219     ourEl: 20       boxHdg: 285
ourAz: 24 ourEl: 20 boxHdg: 285
moveMotors az=24  el=20
Servo write az=24  el=20
CF_VARIO:EA 04 07 FF A8 39 60 00 00 6A 00 00 00 E2 81 
hom.lat:47.4841461 hom.lon:9.5693798 hom.alt:-1 cur.lat:47.4827957 cur.lon:9.5674925 cur.alt:65 cur.alt_ag:66
  hc_vector.az:223.40  hc_vector.el:17.75  hc_vector.dist:206  elapsed:423.510
worldAz: 223     ourEl: 17       boxHdg: 285
ourAz: 28 ourEl: 17 boxHdg: 285
moveMotors az=28  el=17
Servo write az=28  el=17
CF_VARIO:EA 04 07 00 87 98 60 00 00 6A 00 00 00 48 81 
hom.lat:47.4841461 hom.lon:9.5693798 hom.alt:-1 cur.lat:47.4827385 cur.lon:9.5670986 cur.alt:66 cur.alt_ag:67
  hc_vector.az:227.62  hc_vector.el:16.10  hc_vector.dist:232  elapsed:424.070
worldAz: 227     ourEl: 16       boxHdg: 285
ourAz: 32 ourEl: 16 boxHdg: 285
moveMotors az=32  el=16
Servo write az=32  el=16
CF_VARIO:EA 04 07 01 0D 2A 60 00 00 01 00 64 09 48 81 
hom.lat:47.4841461 hom.lon:9.5693798 hom.alt:-1 cur.lat:47.4827309 cur.lon:9.5667009 cur.alt:67 cur.alt_ag:68
  hc_vector.az:232.02  hc_vector.el:14.92  hc_vector.dist:255  elapsed:424.631
worldAz: 232     ourEl: 14       boxHdg: 285
ourAz: 37 ourEl: 14 boxHdg: 285
moveMotors az=37  el=14
Servo write az=37  el=14`

Here the console when I press adjust button before home is set:

`Starting src/main
Version:2.22.28
Setting up Wire I2C: SDA:21, SCL:22
Display support activated: Landscape
64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
ESP32 detected
Target MCU Family:3
ESP32/Variant is Dev Module
Expecting ESPNOW In
Selected protocol is CRSF
Expecting STEPPERS motors
headingsource:2 FC Compass
midFront is assumed to be 90 deg
macToEeprom()
eepromToMac()
Soft MAC address set
Please use adjust button then set midFront position
end of setup()
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x4000d099  PS      : 0x00060b30  A0      : 0x800d7840  A1      : 0x3ffb21d0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ffb833c  A7      : 0x00000000  A8      : 0x00000001  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060b20  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40084aa9  LEND    : 0x40084ab1  LCOUNT  : 0x00000027

Backtrace: 0x4000d096:0x3ffb21d0 0x400d783d:0x3ffb21f0 0x400d4633:0x3ffb2210 0x400d5531:0x3ffb2250 0x400dc599:0x3ffb2290

  #0  0x4000d096:0x3ffb21d0 in ?? ??:0
  #1  0x400d783d:0x3ffb21f0 in MoToStepper::setSpeedSteps(unsigned int) at lib/MobaTools/src/utilities/MoToStepper.cpp:369
  #2  0x400d4633:0x3ffb2210 in buttonEvents() at src/main.cpp:263
  #3  0x400d5531:0x3ffb2250 in loop() at src/main.cpp:1129
  #4  0x400dc599:0x3ffb2290 in loopTask(void*) at C:/Users/DTS/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: 173bf258572bb2ef

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4

Starting src/main
Version:2.22.28
Setting up Wire I2C: SDA:21, SCL:22
Display support activated: Landscape
64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
ESP32 detected
Target MCU Family:3
ESP32/Variant is Dev Module
Expecting ESPNOW In
Selected protocol is CRSF
Expecting STEPPERS motors
headingsource:2 FC Compass
midFront is assumed to be 90 deg
macToEeprom()
eepromToMac()
Soft MAC address set
Please use adjust button then set midFront position
end of setup()`

zs6buj, 2025-05-10:
Thanks for the feedback. I'll get onto this in a day or two

zs6buj, 2025-05-16:
Sorry for the wait. I got onto it this morning, and found a macro error on my part prevented instantiating the MobaTools stepper driver. My scope shows pulses on those pins now.  Let me know how it goes. Using the moba stepper driver is new and I just have been too busy to get back to this. It's beta of course.

dimircc, 2025-05-19:
Hello, big thanks for the fix. I tried it and it works fine. Awesome

zs6buj, 2025-05-20:
Cool. Thanks for the helpful feedback. Closing the issue

closed 2025-05-20 by zs6buj