zs6buj/AntTracker #101: help with the servos

brunoproza, 2025-07-18:
hello everything is fine?

I need help with the antenna servos.

I use an ESP32 Dev Module, with telemetry via ESPNOW.

Everything is working, I receive telemetry and can view it, but I'm having trouble with the servos; they don't move.

It's defined as servos.
//#define STEPPERS
#define SERVOS

It's defined as 180
// #define AZ_SERVO_360 // Means the azimuth servo can point in a 360-degree circle, elevation servo 90 degrees.
// Default (comment out #define above) is 180-degree azimuth and flip over 180-degree elevation.

It's defined to test the servos at startup.
#define TEST_MOTORS // Move servos/steppers through their limits, then try box_hdg every 45 degrees of the circle.

However, when I turn it on, it's not testing the servos (they're in the ports). (32 and 33 are the code's defaults)

Everything is powered by 5V. I've also tested the servo separately with this same power supply.

What could I be doing wrong?

I'm using beta version v2.22.29 via Platformio.

brunoproza, 2025-07-19:
An update

I continued testing the antenna and it seems to be tracking (I haven't tested it with the plane in flight yet), so it seems that only the startup service test isn't working.

I'll continue testing and assembling the case. I'll report back.

zs6buj, 2025-07-20:
Hello Bruno, my apologies for the delayed response.

> I continued testing the antenna and it seems to be tracking (I haven't tested it with the plane in flight yet), so it seems that only the startup service test isn't working.

Thank you for this feedback. From this I understand that the servos are moving in normal operation, but the startup servo test is not happening?

Ok, I flashed v2.22.29 and see the problem. I'll try to fix this later today

brunoproza, 2025-07-20:
I tested it by changing the #define SERVO_SPEED parameter, and it really does get faster when it increases.

Now it's time to finish assembling it and test it on the track.

Thank you again for your help.

closed 2025-07-20 by brunoproza