zs6buj/AntTracker #102: Regarding the problems encountered during use, I would like to ask you some questions…

fox-xian, 2025-07-22:
Hello, I am so glad to find and learn about your project. I tried to replicate it on the c3 super mini yesterday, but I encountered a problem. After I configured config.h as required, it compiled smoothly. I flashed the hardware and it can start normally after a few years, but it seems to only execute the servo motor's power-on self-test. I set the WiFi AP mode in the configuration file. Logically, it should generate a discoverable hotspot signal with the SSID I set after booting. But it didn't. It only executed the servo motor self-test and there was no other action. What is the problem? Then I also need to know that the servo motor I bought has a rotation range of 270 degrees. How can I set it in the config.h file? There is one last question. If I don't want to use the servo motor as a power drive in the future and want to change it to a stepper motor, do I need to buy an additional driver board in addition to the stepper motor? For example, tb6600 or other boards with the same type of chips? The above are the problems I encountered, and I sincerely hope to get your answers. Thank you！

fox-xian, 2025-07-22:
I'm glad to receive your reply!
The circuit board I purchased is the cheaper C3 super mini you mentioned. But the WiFi antenna on my circuit board is not red but white and larger than the red version. The seller said this is the Pro version. I tried to flash the official backpack firmware of ELRS, which can create an AP hotspot that can be discovered. And it is also possible to switch to a universal AP firmware, I don't know if this is caused by different referenced WiFi library files. I think since this circuit board cannot implement WiFi AP mode, I have reordered an ESP32 development board and will receive it tomorrow. I will test it on the new circuit board to see if it can succeed.
[image]
Regarding the angle setting of the servo motor mentioned earlier (the 35KG servo motor I purchased has a 270 degree rotatable range, but it seems that only 360 degree and 180 degree ranges are available in config. h), I don't know how to fill in its angle range in config. h. Should I annotate the entire paragraph in the picture below and use it completely as a 180 degree servo motor? Or do we just need to modify the angle below and set the maximum angle to 269 without any comments? ）
[image]
Do you have any good suggestions on whether additional motor driver boards need to be purchased when replacing with stepper motors in the later stage?

zs6buj, 2025-07-22:
Even though your servo can turn through 270 degrees, you only need 180 degrees. It should then simply ignore commands to move further than that. 

Unfortunately I have to sign off right now, but I'll be back tomorrow

fox-xian, 2025-07-23:
@dimircc 
Without a doubt, this is a feasible solution for you. We should all adjust the code or mechanical structure according to our actual situation. The situation here is that the directional antenna I purchased is a large flat panel antenna used for outdoor bridges, which operates at 5.8GHz and has a gain of 27dBi. But he is too big and not suitable for intense flipping or rapid rotation.

fox-xian, 2025-07-23:
@zs6buj Hey, there's something I forgot to tell you earlier. There is an error when compiling the PIO of my newly purchased ESP32 circuit board. This error is mainly to remind me that the library file for SparkFun_u-blogx_GNSArduino-Library is missing. I downloaded it on GitHub myself and extracted it to the lib folder. After compiling, it was successfully executed.

fox-xian, 2025-07-23:
@zs6buj I noticed during the process of drawing the circuit board that the OLED display pin set in your code overlaps with the pin of the electronic compass.
#define SDA           21        // I2C OLED board and/or Compass
#define SCL           22        // I2C OLED board and/or Compass
I have an M8N with a built-in 5833 electronic compass and six wires. Two of them are positive and negative poles, two are serial port connections for GPS, and the other two are serial port connections for electronic compasses. What should I do with the OLED display if I want to use a combination of GPS and electronic compass to calibrate my local position in this situation?

sosko71, 2025-07-23:
Compass uses an I2C connection through which several different devices can be connected simultaneously.

zs6buj, 2025-07-24:
> [@zs6buj](https://github.com/zs6buj) Hey, there's something I forgot to tell you earlier. There is an error when compiling the PIO of my newly purchased ESP32 circuit board. This error is mainly to [...]

Thank you for this. I'll add it into the lib folder